budryerson/TFMini-Plus-I2C #1: Unknown Escape Sequence

slowtoaccept, 2020-01-11:
Running TFMPI2C_example. Get following compiler warning. Arduino Mega 2560

D:\Users\Jerry\Documents\Arduino\libraries\TFMPI2C\examples\TFMPI2C_example\TFMPI2C_example.ino:127:17: warning: unknown escape sequence: '\302'

         printf( "Temp:%2u\°C", tfTemp);    // display temperature,

Also, stops printing after a few seconds with no data transmitted:
10:21:25.463 -> 
10:21:25.463 -> TFMPlus I2C Library Example
10:21:25.498 -> System reset: Status: I2CWRITE
10:21:25.498 -> Reply: 0 0 0 0 0 0 0 0
10:21:25.498 -> Data: 0 0 0 0 0 0 0 0
10:21:25.498 -> Firmware version: Status: I2CWRITE
10:21:25.498 -> Reply: 0 0 0 0 0 0 0 0
10:21:25.498 -> Data: 0 0 0 0 0 0 0 0

budryerson, 2020-01-13:
Jerry,

Feel free to delete the characters `\°` from `printf( "Temp:%2u\°C", tfTemp);`.
It was presumptuous of me to try to include a degree symbol in that line; and I will delete it myself from future editions.

Did you switch your TFMini-Plus from the default UART mode into I2C mode using the TFMini-Plus library?  Did the **TFMP_example.ino** sketch from that library run correctly while your device was in UART mode?

I apologize for the delay in responding.  Let's get this working.

Bud Ryerson
San Francisco

slowtoaccept, 2020-01-14:
Hi Bud
I am impressed with the thoroughness of your work! 

Yes, I got the TFMP_example.ino up and running after I switched the TX and RX lines! 

After reading further, I discovered that TFMini Plus can only be put into I2C mode issuing a SET_I2C_MODE, while in the UART mode. 

I assume it would go: 
      printf( "Set I2C Mode "); 
     if( tfmP.sendCommand( SET_I2C_MODE, 0)) 
     { 
     printf( "Set I2C.\r\n"); 
     } 
     else tfmP.printErrorStatus(); 

So, two 4-line” .inos”, one to switch to I2C and another to switch back, or factory reset.

The challenge I have is to translate your work into Particle-land. Testing on Arduino Mega, but  I’m using a Particle Boron.

Thanks for your prompt response.

Jerry  

From: Bud Ryerson 
Sent: Monday, January 13, 2020 12:35 PM
To: budryerson/TFMini-Plus-I2C 
Cc: slowtoaccept ; Author 
Subject: Re: [budryerson/TFMini-Plus-I2C] Unknown Escape Sequence (#1)

Jerry,

Feel free to delete the characters \° from printf( "Temp:%2u\°C", tfTemp);.
It was presumptuous of me to try to include a degree symbol in that line; and I will delete it myself from future editions.

Did you switch your TFMini-Plus from the default UART mode into I2C mode using the TFMini-Plus library? Did the TFMP_example.ino sketch from that library run correctly while your device was in UART mode?

I apologize for the delay in responding. Let's get this working.

Bud Ryerson
San Francisco

budryerson, 2020-01-14:
Jerry,

Remember, after getting the device running in UART mode and sending the `SET_I2C_MODE` command, you will then have to physically switch the device connection over to I2C lines to communicate.  Also, the SET_XXX_MODE commands do **not** require a subsequent `SAVE_SETTINGS` command to take effect.

```
printf( "Set I2C Mode ");
if( tfmP.sendCommand( SET_I2C_MODE, 0))
{
    printf( "Set I2C.\r\n");
}
else tfmP.printStatus( false);  // true = show frame data; false =  show command reply
```

A `RESTORE_FACTORY_SETTINGS` command will not restore the device to UART mode.  This is done **only** by sending a `SET_UART_MODE` command... in I2C mode, of course.

It's better, I think, to set the device to one mode or the other and be done with it.  Trying to keep track of what mode the device is in can be very frustrating.

Please advise me of your progress.

Best regards,

Bud Ryerson
San Francisco

slowtoaccept, 2020-01-14:
Hello Bud

Got it. Thanks for the hand holding!

Hope the weather is good in Frisco.

Jerry 

From: Bud Ryerson 
Sent: Tuesday, January 14, 2020 2:30 PM
To: budryerson/TFMini-Plus-I2C 
Cc: slowtoaccept ; Author 
Subject: Re: [budryerson/TFMini-Plus-I2C] Unknown Escape Sequence (#1)

Jerry,

Remember, after getting the device running in UART mode and sending the SET_I2C_MODE command, you will then have to physically switch the device connection over to I2C lines to communicate. Also, the SET_XXX_MODE commands do not require a subsequent SAVE_SETTINGS command to take effect.

printf( "Set I2C Mode ");
if( tfmP.sendCommand( SET_I2C_MODE, 0))
{
    printf( "Set I2C.\r\n");
}
else tfmP.printStatus( false);  // true = show frame data; false =  show command reply
A RESTORE_FACTORY_SETTINGS command will not restore the device to UART mode. This is done only by sending a SET_UART_MODE command... in I2C mode, of course.

It's better, I think, to set the device to one mode or the other and be done with it. Trying to keep track of what mode the device is in can be very frustrating.

Please advise me of your progress.

Best regards,

Bud Ryerson
San Francisco

closed 2020-02-17 by budryerson