budryerson/TFMini-Plus-I2C #2: TFMini-Plus  and TFMini-s

patrickpoirier51, 2020-06-14:
Hello Bud,

Thanks for this excellent code and document, it helps pushing this interesting sensor to new heights... as for Ardupilot for example :-)
https://ardupilot.org/copter/docs/common-benewake-tfmini-lidar.html

Just want to inform you about a little misleading info about the compatibility list:
The TFMini-s has the same controler/firmware as the TFMini PLUS , the only difference is the enclosure that is IP65 on the PLUS.

So TFMini-s AND TFMini Plus can be used with your code.

Best Regards

budryerson, 2020-06-14:
The TFMini-S is apparently a new product from Benewake. I don't have one to play with yet; but from the TFM-S data sheet, I see a problem with "abnormal data values" (p.12).  This library treats distance (dist) and return signal strength (flux) as unsigned integers and does not interpret -1, -2, or -4 results.  The TFM-Plus data sheet (v4) doesn't mention "abnormal values," but I'll bet they're the same.  I'm going to have to look into it.  Thanks, Patrick.
Bud Ryerson, San Francisco

budryerson, 2020-06-14:
P.S. - I like the "ears" on the TFMini-S case.  I find the case of the TFMini-Plus, with its two M2 screw sockets, to be fussy and much less versatile to mount.

closed 2021-02-03 by budryerson

chahatdeep, 2021-06-12:
Hi,
Opening this again. Was this solved? I'm not able to get my TFMINI-S working on UART but my older TFMINI (Not S model) works perfectly with RANGE FINDER TYPE = 20. Am I missing something?

budryerson, 2021-06-12:
Hello,
Starting with version v.1.4.0 (15JUN20) all TFMini-Plus library **data values were changed from unsigned to signed integers** and the library responds to "abnormal" data codes as per the TFMini-S Product Manual to make it more compatible with that product.
```
-  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -
 Distance | Flux (Strength) | Comment
   -1        Other value      Flux (Strength) is less than 100
   -2           -1            Signal strength saturation
   -4        Other value      Ambient light saturation
-  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -  -
```
Does that answer your question?  Please let me know.

Bud Ryerson
San Francisco