budryerson/TFMini-Plus-I2C #4: every second data package being 0

evsc, 2020-07-27:
Thank you for both TFMiniPlus libraries! I am just switching over to the I2C version. 
I noticed that every second data package I receive has 0 values. 

My output looks like this

> Dist:0000cm Flux:00000 Temp:-256°C  
> Dist:0194cm Flux:04201 Temp:65°C  
> Dist:0000cm Flux:00000 Temp:-256°C  
> Dist:0192cm Flux:04199 Temp:65°C  

The sensor status still reports back as TFMP_READY on each reading. 
I tried with a different sensor, and I also tested different measurement+frame rates. 

Is that a behavior you've observed before? 

[Arduino DUE, Arduino 1.8.5 on Ubuntu]

budryerson, 2020-07-27:
Eva,
That's a new one.
Please check your Data Frame Rate to be sure that it matches or exceeds the rate at which you are requesting data.
Also, make sure that v1.4.1 of the library is installed.
And if you can, please send me your code so that I might be able to duplicate the error.
Thanks,
Bud Ryerson

evsc, 2020-07-27:
Hey, thanks for the fast response. 

I installed the library today from github, so it is the most recent one. Also, after running into this trouble, i went back to your example code [TFMPI2C_example.ino](https://github.com/budryerson/TFMini-Plus-I2C/blob/master/example/TFMPI2C_example.ino) and ran it without any changes from my side. Regular Wire bus speed, FRAME_20, and delay(50) in the loop. 

I have used those sensors in UART mode before, where i had set different frame update rates, but I don't imagine those changes would somehow influence the behavior now? 

I'll see if I can test with a different Arduino model.

evsc, 2020-07-28:
Okay, I can now confirm that everything works fine with Arduino UNO and MEGA. 
There are apparently some issues with the Arduino DUE and the Wire library.

budryerson, 2020-10-12:
evsc - Did you resolve the issues with the Arduino Due?  Please let me know.