budryerson/TFMini-Plus-I2C #5: i2c to serial

fffmonkey1, 2020-07-31:
Hi

I discovered that using i2c in arducopter occurs signals spark frequently.  Would you write the code to change tfmini-plus i2c back to serial?

Million thanks

budryerson, 2020-07-31:
I'm sorry to hear that you're having a problem.
In the `setup()` section of the example code, `TFMPI2C_example.ino`, there are several
example commands.  Please add the following command to the list:
```
    // - - Set Serial Mode - - - - - - - - - - -
    printf( "Set Serial Mode: ");
    if( tfmP.sendCommand( SET_SERIAL_MODE, 0))
    {
        printf( "passed.\r\n");
    }
    else tfmP.printReply();
```
In my experience, the TFMini-Plus device will switch immediately
to UART (serial) communication mode.  A following `SAVE_SETTINGS`
command is not necessary.

I hope that helps.  Please let me know.

Bud Ryerson
San Francisco

fffmonkey1, 2020-08-02:
Thank you very much!

closed 2020-10-12 by budryerson