budryerson/TFMini-Plus-I2C #7: Errors on v 1.5.0

judas79, 2020-09-26:
Thanks for creating a I2C library for the TFmini Plus.  I ran your example, for an Uno,  for  TFMPI2C_changeI2C.ino from the v 1.5.0 library and received a couple of errors on this bit of code.
else tfmP.printErrorStatus();
received this error:
'class TFMPI2C' has no member named 'printErrorStatus'; did you mean 'printStatus'?
I tried changing the code to:
else tfmP.printStatus();
received thus error:
'void TFMPI2C::printStatus()' is private within this context
Also, received an error for this:
printlnf( "\r\n*****************************\r\n");
...but I'm guessing printf doesn't support the  new line 'ln' in 'printlnf', so I omitted the 'ln'.

I do have a question.   Did you notice  if the speed, in getting the tfDist data, was superior, using the I2C, to Software Serial?  I'm using the serial method on a 2 axis scanner, as of now, and find it lagging, even set at delay(50);,

budryerson, 2020-09-26:
Oh, my!
1) I meant **`tfmP.printReply()`** in case of a command error, not `tfmP.printErrorStatus();`
2) I meant **`printf()`** not `printlnf()`.  That was a typo.

Thank you for your gentle correction.  I have fixed the code.  I regret my carelessness.

I generally prefer the I2C interface, but I have not done any serious timing tests.  The I2C bus requires a command to send data.  The serial bus is always sending data.  With a hardware serial port, good data could already be in your receive buffer when you request it.  I'm not sure about Software Serial.  I guess I'm also not sure what the 'it' is that you find lagging.  Please tell me more, if you care to.

Bud Ryerson
San Francisco

judas79, 2020-09-27:
Thanks for answering, so promptly:
I caught the typo, but not the ‘tfmP.printErrorStatus();’ error, thanks.
I was weary of using your code, to change my TFmini plus to I2C, because of the error.  I will use it soon.

After seeing a few youtube videos, of people making an x, y ,z, scanner, I decided I had to have one; an inexpensive one. I used the libraries for the TFminiplus, the LCD4Bit_mod library for my cheap Chinese LCD 1602 display shield, with buttons, and the servo library.  Anyhow the sketch runs like a sloth on a pogo stick.
  

As I’m sure know, using the TFmini plus, the quality has to be above a certain threshold, for the tfDist to give the user an accurate measurement in millimeters.  If the reading has a low quality, then we get the tfmP.printFrame();
Its been my experience, that the TFmini will spit out a cluster of quality measurements, then occasionally, not send any data (software serially) that is of quality, and get the tfmP.printFrame()response.  Its almost 5 to one,  bad quality data, between clustering, and since the serial baud is 115200 which is roughly 12kb a second (my calculation may be incorrect), I was thinking I could get a lot more quality responses, using I2C, which is 400kb per second.

If you want to see the code I made, I can email it to you.  I don’t want to post it, while its in this sorry  state, until the bugs are ironed out, but do not mind sharing.

budryerson, 2020-09-28:
Normally, I'd guess you're overrunning your serial receive buffer, but I don't know what you mean by 'bad quality data.'  What Arduino are you using?  In any case, using multiple serial ports (soft or hard) for multiple LiDARs is a bad idea.  That's what the I2C bus is for.

I changed all the `printf()` statements to `Serial.print()` in **TFMPI2C_changeI2C.ino**,   I tried it out this afternoon and it seems to be working okay.  I'll do the same for **TFMPI2C_example.ino** but not today.

I'll look at your code - and pictures too if you've got 'em.   Feel free to email me at: bud-at-budryerson-dot-com

judas79, 2020-09-28:
I'm using an Uno.  I'm also using one Software Serial port with the Software Serial library, for the TFmini P., and the the hardware serial to send the gathered data to the Processing sketch.
As I understand the doc. I read, for the tfminiP library, the TF sensor takes 3 reading;  distance, quality (the quality of the read itself) and temperature (internal temp. of the TF unit).  If the quality of the read is above a certain threshold, we get the distance read in millimeters, if it doesn't reach the quality amount threshold, it will give us tfmP.printFrame();  which is the raw data from the read in hexadecimal.
I am curious, there seems to be a lot of sketches, out there using printf., what are the advantages for using it?
I will email you, thankyou.

budryerson, 2020-09-28:
FYI, there is also a `getData( dist);` command available that passes back only the distance data.

If the distance value is -1 or -4 or the flux is -1, the library throws an error.  You don't have to respond to the error. The `else` statement after `getData();` in the example code is just for troubleshooting.

`printf();` is much more versatile and compact than `Serial.print();`.  It's even included in the IDE for some Adriunos.  But it has one large, annoying problem.

This is the repository for the TFMPlus-I2C library.  If we are having trouble with the TFMPlus in serial mode, we should move the conversation over to there.  I will close out this thread now.

closed 2020-09-28 by budryerson

judas79, 2020-12-04:
Thanks for answering, so promptly.  I don’t know if I am supposed to be replying to this via email, or if I should use the GitHub site, so let me know if you receive this email, please.
I caught the typo, but not the ‘tfmP.printErrorStatus();’ error, thanks.

I was weary of using your code, to change my TFmini plus to I2C, because of the error.  I will use it soon.

After seeing a few youtube videos, of people making an x,y,z, scanner, I decided I had to have one; an inexpensive one. I used the libraries for the TFminiplus, the LCD4Bit_mod library for my cheap Chinese LCD 1602 display shield, with buttons, and the servo library.  Anyhow the sketch runs like a sloth on a pogostick.
  

As I’m sure know, using the TFmini plus, the quality has to be above a certain threshold, for the tfDist to give the user an accurate measurement in millimeters.  If the reading has a low quality, then we get the tfmP.printFrame();
Its been my experience, that the TFmini will spit out a cluster of quality measurements, then occasionally, not send any data (software serially) that is of quality, and get the tfmP.printFrame()response.  Its almost 5 to one bad quality data, between clustering, and since the serial baud is 115200 which is roughly 12kb a second (my calculation may be incorrect), I was thinking I could get a lot more quality responses, using I2C, which is 400kb per second.

If you want to see the code I made, I can email it to you.  I don’t want to post it, while its in this sorry  state, and the bugs are ironed out, but do not mind sharing.

From: Bud Ryerson
Sent: Saturday, September 26, 2020 4:00 PM
To: budryerson/TFMini-Plus-I2C
Cc: Roddy; Author
Subject: Re: [budryerson/TFMini-Plus-I2C] Errors on v 1.5.0 (#7)

Oh, my!
1. I meant tfmP.printReply() in case of a command error, not tfmP.printErrorStatus();
2. I meant printf() not printlnf(). That was a typo.
Thank you for your gentle correction. I have fixed the code. I regret my carelessness.
I generally prefer the I2C interface, but I have not done any serious timing tests. The I2C bus requires a command to send data. The serial bus is always sending data. With a hardware serial port, good data could already be in your receive buffer when you request it. I'm not sure about Software Serial. I guess I'm also not sure what the 'it' is that you find lagging. Please tell me more, if you care to.
Bud Ryerson
San Francisco
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

judas79, 2020-12-05:
Yes, I’ve been using this to get my data:
    mySerial.listen();
    tfmP.getData( tfDist);

instead of the conditional statement:
if( tfmP.getData( tfDist, tfFlux, tfTemp))
{ }

I was attempting to get the data, when its available from the serial at the same time I obtained the servo x, y coordinates; 
 I also omitted the else statement:

else                  // If the command fails...
    {
      //tfmP.printFrame();  // display the error and HEX dataa
    }
For, as you said, it seemed to be for troubleshooting.  It may very well be the code I wrote, that is the core of my projects problem.  I’m still learning…
I realize we got off track, and started talking about using the serial TFmini library, and still am going to try out the I2C library, I just wanted to give the background of what I tried prior to coming here.

Thank you again for all your input

From: Bud Ryerson
Sent: Monday, September 28, 2020 6:14 PM
To: budryerson/TFMini-Plus-I2C
Cc: Roddy; Author
Subject: Re: [budryerson/TFMini-Plus-I2C] Errors on v 1.5.0 (#7)

FYI, there is also a getData( dist); command available that passes back only the distance data.
If the distance value is -1 or -4 or the flux is -1, the library throws an error. You don't have to respond to the error. The else statement after getData(); in the example code is just for troubleshooting.
printf(); is much more versatile and compact than Serial.print();. It's even included in the IDE for some Adriunos. But it has one large, annoying problem.
This is the repository for the TFMPlus-I2C library. If we are having trouble with the TFMPlus in serial mode, we should move the conversation over to there. I will close out this thread now.
—

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

budryerson, 2020-12-06:
I've been looking at your `TFMP_LCD4BitMod_Servos_3_cleaned.ino` file.
The main loop doesn't seem to do much except look for a key press, and I am not clear about what the different key presses are supposed to do.
The `startLidar()` function appears to be the heart of the operation, and it looks like it would produce one 27 minute long scan.
There seems to be an awfully large number of `15ms` and `50ms` delays in the code. The purpose of all of these delays is not clear to me.
Is it possible that we are missing a close curly brace "}" around line 252?
Does this compile and execute correctly?

Thanks for sharing,

Bud Ryerson
San Francisco

judas79, 2020-12-06:
First, thankyou very much for looking at my sketch.  I’ve been stuck for weeks, and have not progressed since that sketch.  I’m including a pic and short description of my setup, even though the wires from the pic are hooked up to another Arduino, which I was using to change the TF mini Plus from serial to I2C, using your sketch.  Your sketch,  did find the old address, but I didn’t react to whatever it was that I was supposed to do next, in a timely manner.  It may have been, to enter y for yes, but its been a long time, and don’t recall.  I mention this to relay that my TF mini plus is still in serial mode. I’m using a combined 16x2 TFT and 6 buttons Chinese TFT shield, Arduino Uno, proto screw shield, 2 servos and the TF Mini Plus.  I tried to combine the code, that I could find, for all the individual components, into 1 sketch.   “The main loop doesn't seem to do much except look for a key press, and I am not clear about what the different key presses are supposed to do.“The buttons,  seemed to me, to have to be  executed first.  When I tried calling them, from there own function it would not execute, I assume, because other things where now also running, and the buttons when pressed, would not interrupt, the flow of the code.   I could, aggressively and repeatedly push the button many times and sometimes get the button to change state.  I only actually have been using 2 buttons, the reset button and the start button, that starts the sketch.Previously I used another button to stop the scan, but got rid of that, because it did not do what I wanted it to do;  stop the scan when pressed, one time   BTW, before I can press start, I have to load  up a processing sketch, not written by me, that will draw the scan coordinates as dots.  This Prossesing program uses  the hardware serial pins.(Prossesing sketch attached) “There seems to be an awfully large number of 15ms and 50ms delays in the code. The purpose of all of these delays is not clear to me.”Originally, I had all those delays, all set to 15ms.  Most of, if not all those delays where in the code, that I derived my TFMP_LCD4BitMod_Servos_3_cleaned.ino from.  In my assessment of the problem with the servos pausing, and shaking, I guessed that perhaps the problem was that there wasn’t enough time, to execute each sample taken by the TF mini Plus.  I had read somewhere, that the TF mini serial works at 100mhz and setting the delay to 50ms would approximate that frequency.  I tried many other delay values, and they did not change the shakes of the servos either. “Is it possible that we are missing a close curly brace "}" around line 252?“It compiled and ran, but I would not say executed correctly;  the shaking starts about halfway through the scan.  I will look for the missing brace, thanks!.“Does this compile and execute correctly?“I’ve been discouraged from continuing trying, on this project, because I was stuck and out of ideas.  You have given me hope, that this project has a solution, with some outside help.  Sent from Mail for Windows 10 From: Bud RyersonSent: Saturday, December 5, 2020 8:09 PMTo: budryerson/TFMini-Plus-I2CCc: Roddy; AuthorSubject: Re: [budryerson/TFMini-Plus-I2C] Errors on v 1.5.0 (#7) I've been looking at your TFMP_LCD4BitMod_Servos_3_cleaned.ino file.The main loop doesn't seem to do much except look for a key press, and I am not clear about what the different key presses are supposed to do.The startLidar() function appears to be the heart of the operation, and it looks like it would produce one 27 minute long scan.There seems to be an awfully large number of 15ms and 50ms delays in the code. The purpose of all of these delays is not clear to me.Is it possible that we are missing a close curly brace "}" around line 252?Does this compile and execute correctly?Thanks for sharing,Bud RyersonSan Francisco