budryerson/TFMini-Plus-I2C #8: I2C bus locked...

ZorchDiabetes, 2020-10-24:
Hi Bud, 

Everything has been working fine, and then I somehow got it into I2C-WRITE:

attempt to reset I2C bus
Starting I2C bus recovery
bus recovery done, return in 2 secs
attempt to reset I2C bus
Starting I2C bus recovery
bus recovery done, return in 2 secs

TFMPlus I2C Library 1.5.0
System reset:  Status: I2C-WRITE 00 00 00 00 00 00 00 00
Firmware version:  Status: I2C-WRITE 00 00 00 00 00 00 00 00
Data-Frame rate:  Status: I2C-WRITE 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00

Where / how should I recover the bus?

In your TFMPI2C_example.ino, I added these lines, to no avail:

void setup()
{
    Serial.begin( 9600);   // Initialize terminal serial port
    printf_begin();          // Initialize printf library.
    delay(20);
    
    **Serial.println("attempt to reset I2C bus");
    tfmP.recoverI2CBus(27, 28);**
    
    Wire.begin();            // Initialize two-wire interface
//    Wire.setClock( 400000);  // Set I2C bus speed to 'Fast' if
                               // your Arduino supports 400KHz.
    **Serial.println("attempt to reset I2C bus");
    tfmP.recoverI2CBus(27, 28);**
    
    printf( "\r\nTFMPlus I2C Library 1.5.0\r\n");  // say 'hello'

    // Send some example commands to the TFMini-Plus
    // - - Perform a system reset - - - - - - - - - - -
    printf( "System reset: ");
    if( tfmP.sendCommand( SOFT_RESET, 0))
    {
        printf( "passed.\r\n");
    }

The board is a pro mini with a Lora radio, https://www.makerfabs.com/wiki/index.php?title=Maduino_Lora_Radio_(433M/868M)

Regards,

Lane

budryerson, 2020-12-31:
I'm sorry I missed this -  and by now I'm probably much too late.  Are you still having a problem?
Resetting the I2C bus is for the processor not the sensor.  If the sensor is in I2C mode, you will have to send the `SET_SERIAL_MODE` command via the I2C bus to get it back into Serial mode.
Again, I apologize for missing your message.
Let me know if I can help.
Budf

budryerson, 2021-01-24:
On January 1, 2021, ZorchDiabetes wrote: "Thanks for getting back to me. I was able to get the TFmini working again - when I bricked it with I2C, it reverted to serial (factory reset)? I used a Mega to get it back into I2C mode before reconnecting it to the LORA, which only has one serial port... The project was quite fun; I learned a lot."

closed 2021-01-24 by budryerson