budryerson/TFMini-Plus-I2C #11: Change mode from UART to I2C!

tusharvb19, 2021-01-30:
Hi, I did try to change the mode form UART to I2C by following some guide
Help needed.
Here is the error screen snip!
[image]

budryerson, 2021-02-03:
The Benewake GUI works only in Serial Mode.  If you sent a command to change to I2C mode, and the command was successful, then you can only communicate in I2C mode.  An Arduino Pro Mini or Mega 2560 is not expensive and has an I2C interface.  If you get an Arduino, I can help you.  We will make this work.

Givy, 2022-01-20:
Hi Bud,

Do you know if there is a GUI interface connection for TFMini-S to use it when its already in the I2C mode to change the speed from 100 to 400 or set it back to the factory setting etc.?

Thanks 
Givi

budryerson, 2022-01-20:
Hey Givi,
Thanks for writing. The TFMini-S behaves as an I2C Slave devices and bus speed is set by the Master, the Arduino in this case. Use the `Wire.setClock( clockFrequency)` function to change the Artduino bus speed.
I hope I got that right and that it answers your question.
Bud

Givy, 2022-01-20:
Hi Bud,
Thanks for your prompt response and info.  I went to their web site "http://en.benewake.com/download" and found some GUI for updating and testing the TFmini-plus in UART mode (no idea what the latest firmware version is or where to get it).  
So, in I2C mode basically, we work with it by using the "Wire" library!  Please let me know.

Regards,
Givi

budryerson, 2022-01-21:
Givi,

Whenever I wonder, "Gee, did I blow this thing up?  Does it still work?" I check to see if the internal red light is on.  If it is, the chances are that the device still works.  The next step, for me, is the manufacturer's GUI, which will tell me if I'm getting data.  That is **ALL** I have ever used that GUI for.  After that, for me, it's the "great" TFMPI2C library.  The example sketch `TFMPI2C_simple.ino` is as simple as I could make it.  From there, you can add commands, such as checking the firmware version.

Documentation for the TFMini-S is available here:
http://en.benewake.com/res/wuliu/docs/15936159693952399SJ-PM-TFmini-S A01 Product Manual.pdf
http://en.benewake.com/res/wuliu/docs/15936159500205286SJ-PM-TFmini-S-01 A01 Datasheet.pdf

For the latest firmware and firmware updater software you need to contact Benewake support.  They will provide you with a link.

Have fun,

Bud

Givy, 2022-01-23:
Hi Bud,

Thanks again.  You are great.  It is working for me, and I actually have two of them running simultaneously, but one is showing 2cm shorter than the other one, don't know why.  That’s why I thought there could be a way to adjust/modify the accuracy in code or firmware etc.
Regards,
Givi

budryerson, 2022-01-23:
Givi,
The manual states that accuracy is ±6cm at distances between 0.1 and 6 meters, then ±1% of the distance at 6 - 12 meters.  Repeatability is "directly related to the Strength value measured and the output frame rate."  However, if your error is consistent, then you can easily compensate in your code.  Also, be sure they're pointing at the same target. 😄
Bud

Givy, 2022-01-24:
😄

closed 2022-03-06 by budryerson