budryerson/TFMini-Plus-I2C #12: Compiler errors when using platformio (and probably others)

hb020, 2021-02-07:
This happens on both the latest version from Arduino IDE libraries (1.5, 03 SEP 2020), and the latest git version (also 1.5, but 27 SEP 2020)

There are 2 errors like this:
```
<redacted>/Libraries/TFMPI2C/src/TFMPI2C.cpp: In member function 'bool TFMPI2C::getData(int16_t&, int16_t&, int16_t&, uint8_t)':
<redacted>/Libraries/TFMPI2C/src/TFMPI2C.cpp:97:55: error: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second: [-Werror]
Compiling .pio/build/esp12e/FrameworkArduino/Esp-frag.cpp.o
     Wire.requestFrom( (int)addr, TFMP_FRAME_SIZE, true);
                                                       ^
In file included from <redacted>/Libraries/TFMPI2C/src/TFMPI2C.cpp:72:0:
<redacted>/.platformio/packages/framework-arduinoespressif8266/libraries/Wire/Wire.h:71:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int, int)
     uint8_t requestFrom(int, int, int);
             ^
<redacted>/.platformio/packages/framework-arduinoespressif8266/libraries/Wire/Wire.h:65:12: note: candidate 2: size_t TwoWire::requestFrom(uint8_t, size_t, bool)
     size_t requestFrom(uint8_t address, size_t size, bool sendStop);
            ^
```

Best is to address a specific signature of TwoWire::requestFrom, like I did here.
I can make a pull request for that if you want.

```
diff --git a/src/TFMPI2C.cpp b/src/TFMPI2C.cpp
index 6688da6..4a1aee2 100644
--- a/src/TFMPI2C.cpp
+++ b/src/TFMPI2C.cpp
@@ -94,7 +94,7 @@ bool TFMPI2C::getData( int16_t &dist, int16_t &flux, int16_t &temp, uint8_t addr
     // - - - - - - - - - - - - - - - - - - - - - - - - - - - -
     // Request one data-frame from the slave device address
     // and close the I2C interface.
-    Wire.requestFrom( (int)addr, TFMP_FRAME_SIZE, true);
+    Wire.requestFrom( (uint8_t)addr, (size_t)TFMP_FRAME_SIZE, true);

     memset( frame, 0, sizeof( frame));     // Clear the data-frame buffer.
     for( uint8_t i = 0; i < TFMP_FRAME_SIZE; i++)
@@ -253,7 +253,7 @@ bool TFMPI2C::sendCommand( uint32_t cmnd, uint32_t param, uint8_t addr)

     // Request reply data from the device and
     // close the I2C interface.
-    Wire.requestFrom( (int)addr, (int)replyLen, true);
+    Wire.requestFrom( (uint8_t) addr, (size_t)replyLen, true);

     memset( reply, 0, sizeof( reply));   // Clear the reply data buffer.
     for( uint8_t i = 0; i < replyLen; i++)
```

budryerson, 2022-03-02:
OMG.  Was that a year ago?!?  I apologize for not responding.  I meant to get back to this and then forgot about it.

I see what you mean.  My intention was to re-type `addr` from `uint8_t` to `int` before sending it to the Wire library because the `int` type is what the Wire library is expecting.

If I understand you correctly, the code compiles successfully as is, but you are getting notifications from your "PlatformIO" IDE that this may not be a C++ best practice.  As a solution, you suggest leaving `addr` as a type `uint8_t` and letting the Wire library do the retyping.

I do not know how to answer your question.  I have not gotten any errors myself, and I regret having to say that I do not know what a "pull request" is; but if it makes your process easier or better, then I think that you should do it.

Please let me know what you decide; and thank you for taking an interest in the library.

Bud Ryerson
San Francisco

hb020, 2022-03-03:
Did some more digging, adapted my proposal to match more versions of the Wire library. See the Pull request.

budryerson, 2022-03-04:
Now I see that the problem is not retyping the I2C address.  The problem is using a boolean value `true` for the I2C stop bit.  But instead of retyping the boolean value `int(true)` as you proposed, I think the problem can be resolved simply by replacing the boolean value with a literal value `1`.
`Wire.requestFrom( (int)addr, TFMP_FRAME_SIZE, 1);`
What do you think?

hb020, 2022-03-04:
should be OK as well.

budryerson, 2022-03-06:
Hi.  I deeply appreciate your gentle corrections and I welcome your collaboration.  For now, I'm going to close this conversation while I try to integrate your suggestions.  In the meantime, can you please tell me more about yourself?  I would like to be able to give you credit for your contributions.
Thanks,
Bud

closed 2022-03-06 by budryerson

hb020, 2022-03-11:
> In the meantime, can you please tell me more about yourself? 

PM sent via Mail a couple of days ago.

Kind regards