budryerson/TFMini-Plus-I2C #13: No distance or temperature is displayed via I2c

Lordcybertracker, 2021-08-28:
Hi there,
I have the problem that it doesn't send me a distance.
The Seriall Log says:

```
TFMPlus I2C Library 1.5.1
System reset: 
[20:36:25.057] passed.
Firmware version: 
[20:36:25.553] 2.1.5
Data-Frame rate: 
[20:36:26.065] 20Hz.
[20:36:26.561] Dist:0000cm Flux:00000 Temp:-256Â°C
[20:36:26.608] Dist:0000cm Flux:00000 Temp:-256Â°C
[20:36:26.672] Dist:0000cm Flux:00000 Temp:-256Â°C
[20:36:26.721] Dist:0000cm Flux:00000 Temp:-256Â°C
[20:36:26.768] Dist:0000cm Flux:00000 Temp:-256Â°C
[20:36:26.833] Dist:0000cm Flux:00000 Temp:-256Â°C
```

TF Mini Plus is set to I2C. Is also recognized.

What do I have to change there?
thanks

budryerson, 2021-08-28:
Well, that's not right.   Please send me the code that you're using to get this output so I can take a look at it.
Which serial monitor software are you using?
And, if it's not too much trouble, can you include pictures of your setup?  It helps me.
Thanks,
Bud Ryerson
San Francisco

Lordcybertracker, 2021-08-29:
Thanks very much.

Image of the setup will not bring you anything. Because it is an own board.

Based on an Atmega 1284p with 8Mhz crystal and Mightycore bootloader.

 

I have VCC red on a 5V laboratory power supply, GND black together on the Arduino and laboratory power supply, green SCL on the SCL of the board, white SDA on the SDA of the board.

 

Each with 4.7k pullups

 

In the i2C scanner he finds with the Snesor under 0x10.

 

Even if I remove the lidar, the code works.

 

I've attached code.

 

Monitor I use the Arduino integrated or the Serial Port Utility for both the same.

 

Many thanks for the help.

Lordcybertracker, 2021-08-29:
I switched to I2c with these commands:
5A 05 0A 01 6A 
5A 04 11 6F
Is that maybe the fault?

 

Sorry for the clarity. Because I have to work with a translator.

budryerson, 2021-08-29:
You wrote, "I've attached code."  I did not receive your attachment.
Did the TFMini-Plus work in UART mode?
Please be patient.  We will fix this.  I'm certain.

Lordcybertracker, 2021-08-29:
Before switching to I2C, it ran in the Uart. with the tool.

I used the original code from you without changes because the I2C address has remained and is also recognized.

Lordcybertracker, 2021-08-29:
[TFMPI2C_example.zip](https://github.com/budryerson/TFMini-Plus-I2C/files/7072044/TFMPI2C_example.zip)

Lordcybertracker, 2021-08-29:
[image]

Lordcybertracker, 2021-08-29:
Would be great if we can do that.
As this will be 10 Lorawan flood sensors.
[image]
[image]

budryerson, 2021-08-29:
Which Arduino board do you select in the Arduino IDE?
Did this HSN-TTN PCB of yours work correctly with your TFMijni-Plus in UART mode?
Does this PCB work correctly with other I2C slave devices?
Can you communicate with your TFMini-Plus in I2C mode using a real Arduino?
I still would like to see pictures of what you're doing.
Please let me know.
Thanks,
Bud

Lordcybertracker, 2021-08-30:
Hi there,
I often use the board without any problems.
I have now connected the TFMINI PLus to a NANO board with the same result but without pullups.

How do I reset the lidar to Uart?

That I can test it again there?

thanks

budryerson, 2021-08-30:
I'm going to number my questions to help keep track.
  1. Did you try swapping the SDA/SCL leads?
  2. You wrote: "In the i2C scanner he finds with the Snesor under 0x10."  What I2C scanner are you using?
  3. Please turn off the timestamp in the Arduino IDE Serial Monitor and send me a copy of the output that you get from your Arduino Nano.
  4. Please send me pictures of your connections.
  5. Benewake says that on power up, the device will be in UART mode for about 10 seconds before it switches to I2C.  Did you try that?
And please be patient.  The problem will turn out to be something easy and obvious.  It always does.
Bud Ryerson
San Francisco

Lordcybertracker, 2021-08-31:
Since the sensor didn't give anything else apart from the 0x10, I sent it back as a defect.

The new one comes today or tomorrow.

I'm going to number my questions to help keep track.

1.	Did you try swapping the SDA/SCL leads?
Yes, I have

2.	You wrote: "In the i2C scanner he finds with the Snesor under 0x10." What I2C scanner are you using?
That was the Arduino pattern code.

3.	Please turn off the timestamp in the Arduino IDE Serial Monitor and send me a copy of the output that you get from your Arduino Nano.
Can I do with the new sensor.

4.	Please send me pictures of your connections.
Can I do with the new sensor.

5.	Benewake says that on power up, the device will be in UART mode for about 10 seconds before it switches to I2C. Did you try that?
And please be patient. The problem will turn out to be something easy and obvious. It always does.

I re-energized it several times. But it did not start in serial mode. That the sensor only does 10 seconds UArt is a problem. Because the circuit is battery operated. and after sending, disconnects the power via Mosfet.

Do I actually need pullups for the sensor or are they integrated?

budryerson, 2021-08-31:
You wrote, "Since the sensor didn't give anything else apart from the 0x10..."  I do not understand what you mean.  In what manner did the sensor "give" the 0x10?  When I asked, "What I2C scanner are you using?" you wrote, "That was the Arduino pattern code."  Again, I do not understand what you mean.  What is an Arduino pattern code?  Where did this code appear?

I have used several different LiDARs, as many as three on the same I2C bus at a time, and a variety of Arduinos and Raspberry Pis.  Sometimes I use an oscilloscope to check the data waveform.  So far, I have not needed a pull-up.

Your description of battery operation is mystifying.  I expect the LiDAR to be powered by the USB-Serial converter that you use to run the Benewake GUI test software that you used to send the serial commands.

I hope the new LiDAR works well for you and the High Blackforest Network.

Bud Ryerson
San Francisco

Lordcybertracker, 2021-08-31:
Hi there,

in the Arduino IDE there are example codes for I2C scanners.

 

int nDevices = 0;

 

  Serial.println("Scanning...");

 

  for (byte address = 1; address < 127; ++address) {

    // The i2c_scanner uses the return value of

    // the Write.endTransmisstion to see if

    // a device did acknowledge to the address.

    Wire.beginTransmission(address);

    byte error = Wire.endTransmission();

 

    if (error == 0) {

      Serial.print("I2C device found at address 0x");

      if (address < 16) {

        Serial.print("0");

      }

      Serial.print(address, HEX);

      Serial.println("  !");

 

      ++nDevices;

    } else if (error == 4) {

      Serial.print("Unknown error at address 0x");

      if (address < 16) {

        Serial.print("0");

      }

      Serial.println(address, HEX);

    }

  }

  if (nDevices == 0) {

    Serial.println("No I2C devices found\n");

  } else {

    Serial.println("done\n");

  }

  delay(5000); // Wait 5 seconds for next scan

 

 

With this I get the 0x10 as feedback.

But the sensor did not do more.

 

Of course, I first tested the lidar on the USB adapter. At the beginning he went via UART also via Benewake GUI.

 

They said he's 10 seconds in the UArt. but even there he did not react.

 

That's why I'm thinking of a defective one.

budryerson, 2021-08-31:
Thank you.  Now I understand.  If the I2C scanner sketch reported an address for your LiDAR, that means the LiDAR is correctly connected and working.  If the I2C scanner sketch works, the TFMPI2C_example sketch should work.
Believe me.  I have been where you are.  And in my experience, once the device is set in I2C mode, the only way back to UART mode is by a command in I2C mode.  Benewake claims the 10 second UART window works, but it has never worked for me.  I thought you could give it a try.
This will work.  I am 100% confident that we will figure it out.  But patience and persistence, kindness and a slightly weird sense of humor are required.
Good luck.
Bud

Lordcybertracker, 2021-09-01:
Good Morning,

the new ensor is here now. This works at least with the tool.

So in Uart mode.

I suspect that I made a mistake while switching to I2C.

And that's why there was nothing left.

 

I also come with the instructions that are not clear.

Could you show exactly how to switch to I2C with the Benewake tool?

 

It should be "5A 05 0A MODE SU".

But what IS "MODE" and "SU"?

thanks

budryerson, 2021-09-02:
Please try using the TFMini-Plus library and the TFMP_example sketch to communicate in UART mode, and let me know whether that worked.
The command structure is illustrated on page 17 of the product manual.  Communication MODE (UART/I2C) is the payload for that particular command and SU is the Checksum byte.
[TFmini Plus A05 Product Manual.pdf](https://github.com/budryerson/TFMini-Plus-I2C/files/7101146/TFmini.Plus.A05.Product.Manual.pdf)
[image]

Lordcybertracker, 2021-09-03:
I tested it with Soft Serial.
Because my 2nd serial pins are occupied.
I have attached code.
result:
07:17:19.130 -> TFMPlus Library Example - 19MAY2021
07:17:19.130 -> System reset: Status: TIMEOUT
07:17:20.151 ->  3E 09 10 C2 FE 00 00 00
07:17:20.661 -> Firmware version: Status: TIMEOUT
07:17:21.685 ->  13 00 49 09 10 C2 DF 00
07:17:21.685 -> Data-Frame rate: Status: TIMEOUT
07:17:22.667 ->  00 4E 09 02 C2 DE 00 00
07:17:24.206 -> Status: HEADER
07:17:24.206 -> Data: 58 99 13 00 59 09 00 C2 E1
07:17:25.279 -> Status: HEADER
07:17:25.279 -> Data: E1 58 D9 4E 00 96 04 08 89

If I turn in series, nothing arrives so it should fit.

[TFMP_example.zip](https://github.com/budryerson/TFMini-Plus-I2C/files/7103470/TFMP_example.zip)

budryerson, 2021-09-04:
Thank you for including the code and your output.
It appears that Software Serial is not working.
What board are you using?  Is Software Serial compatible with the board you are using?
Please try using a hardware serial connection.

Lordcybertracker, 2021-09-05:
Good Morning,
I have now tested it with an Arduino UNO and it works.

My Board Baserit on:
https://github.com/MCUdude/MightyCore
With 1284p CPU at 8Mhz.

Can it be due to the 8Mhz?

I cannot use hardware serial on my board.

See wiring diagram picture.
[image]

budryerson, 2021-09-05:
"I have now tested it with an Arduino UNO and it works."
I understand that to mean that an Arduino Uno, using the TFMini-Plus library, can send commands and receive good data.  Please send me a copy of your output.  If you can, please turn off the timestamp.
And, as I always ask, please send me pictures of your setup.
Thanks,
Bud

Lordcybertracker, 2021-09-11:
So finally it works.

it was probably a mistake in the change to I2C mode.

20:23:49.002 -> TFMPlus I2C Library 1.5.1

20:23:49.002 -> System reset: passed.

20:23:49.514 -> Firmware version: 2.1.5

20:23:49.978 -> Data-Frame rate: 20Hz.

20:23:51.011 -> Dist:0155cm Flux:10876 Temp:59°C

20:23:51.480 -> Dist:0155cm Flux:10877 Temp:59°C

20:23:51.993 -> Dist:0155cm Flux:10877 Temp:59°C

20:23:52.504 -> Dist:0155cm Flux:10876 Temp:59°C

20:23:53.017 -> Dist:0155cm Flux:10873 Temp:59°C

20:23:53.532 -> Dist:0155cm Flux:10868 Temp:59°C

20:23:54.043 -> Dist:0155cm Flux:10872 Temp:59°C

20:23:54.505 -> Dist:0155cm Flux:10872 Temp:59°C

20:23:55.017 -> Dist:0015cm Flux:10321 Temp:59°C

20:23:55.532 -> Dist:0016cm Flux:10207 Temp:59°C

20:23:56.047 -> Dist:0015cm Flux:10438 Temp:58°C

20:23:56.510 -> Dist:0015cm Flux:10414 Temp:58°C

 

However, it does not switch on my power supply in the test program.

Code:

const int VEXT = 19; // Digital pin 19 / PC3, power supply LM, I2C

 

 

void setup ()

{

// Power on

  pinMode (VEXT, OUTPUT);

  digitalWrite (VEXT, HIGH);

For some reason it doesn't switch the pin high.

Any idea what that might mean?

 

THX

budryerson, 2021-09-11:
"So finally it works."  That's great news!  I'm happy to hear it.<br />
"However, it does not switch on my power supply in the test program."  From the code you provided, that it does not appear to be a problem associated with this library.<br />
Good luck with the High Black Forest Network,<br />
Bud Ryerson
San Francisco

closed 2021-09-11 by budryerson