budryerson/TFMini-Plus-I2C #14: Wrong constant definition for ENABLE_OUTPUT and DISABLE_OUTPUT

suratkw, 2021-09-06:
Hi, I took me hours why I got range, flux to be all zeros and temperature to be -256. I even checked with an infrared camera and found that there is no light shining out of the sensor.

It turns out that the definition for command ENABLE_OUTPUT and DISABLE_OUTPUT in TFMPI2C.h are swapped.

-- Current
```
#define    ENABLE_OUTPUT              0x00070505  
#define    DISABLE_OUTPUT             0x01070505
```

-- Should be
```
#define    ENABLE_OUTPUT              0x01070505 
#define    DISABLE_OUTPUT             0x00070505
```

By calling ENABLE_OUTPUT command inadvertently disable the output and thus receive all zero output.

budryerson, 2021-09-07:
You are correct.  I changed the definitions in `TFMPI2C.h`, `TFMPlus.h` and `tfmplus.py` and will test the libraries this week.  The Arduino web crawler picks up new releases on Sunday, so the new versions should be in the IDE Library Manager on September 12th.
That was a bad mistake and I feel terrible about it.  Clearly I should have tested everything more thoroughly.  I am very sorry that you had to find out the hard way.
Thank you so much for your help.
Bud Ryerson
San Francisco

P.S. - You don't need an infra-red camera to see if the emitter is on.  It glows visibly when power is applied (and the device is enabled).
On page 21 of the Product Manual, **Sec. 9 (3) No data output** it says:
> **Troubleshooting:** Check whether the power supply is normal; check whether the voltage is within rated voltage range. If power supply is normal, there will be a red light inside the transmitting lens of TFmini Plus.

suratkw, 2021-09-07:
Hi, no problem. Thank you very much for your work on this very useful library.
-Surat

closed 2021-09-11 by budryerson