budryerson/TFMini-Plus-I2C #15: Cannot set framerate

ronangaillard, 2021-09-15:
Hi,

I tried setting the framerate manually with the Arduino Wire library using the following (dirty) code 👍 

```
uint8_t cmndLen = 5;
    uint8_t cmndData[5] = {0x5a, 0x06, 0x03, 0xe8, 0x03};
    uint16_t chkSum = 0;

    for(uint8_t i = 0; i < cmndLen; i++) chkSum += cmndData[i];

    Wire.beginTransmission(deviceAddress);
    Serial.println("sending");
    for(uint8_t i = 0; i < cmndLen ; i++) {
      Serial.println(cmndData[i], HEX);
      Wire.write(cmndData[i]); 
    }
    Wire.write(chkSum);
    Serial.println(chkSum, HEX);
    
    if (Wire.endTransmission(false) != 0) {
      Serial.println("error setting tfmini frame rate");
        return (false); //Sensor did not ACK
    }

    Serial.println("receiving");
    Wire.requestFrom(deviceAddress, (uint8_t)6); //Ask for 6 bytes

    if (Wire.available())
    {
        for (uint8_t x = 0 ; x < 6 ; x++)
        {
          uint8_t incoming = Wire.read();
  
          Serial.println(incoming, HEX);
        }
    }
    else
    {
        Serial.println("No wire data avail");
        return (false);
    }
```

But it did not work as the response I got from the TFMini Plus was always a bunch of `0`.

But I know that the communication works well as I'm able to retrieve the measured distance through I2C.

So I gave a try to your library with the following code : 

```
 Serial.print( "TFMini-Plus reset: ");
        if( tfmP.sendCommand( SOFT_RESET, 0, _tfmini_adress))
        {
            Serial.println( "passed.\r\n");
        }
        else tfmP.printReply();
        
        if( tfmP.sendCommand( GET_FIRMWARE_VERSION, 0, _tfmini_adress))
        {
            Serial.print(tfmP.version[ 0]); // print three single numbers
            Serial.print("'");
            Serial.print(tfmP.version[ 1]); // each separated by a dot
            Serial.print("'");
            Serial.println( tfmP.version[ 2]);
        }
        else tfmP.printReply();

        Serial.print( "Data-Frame rate: ");
        if( tfmP.sendCommand( SET_FRAME_RATE, FRAME_20, _tfmini_adress))
        {
            Serial.print( FRAME_0);
            Serial.println("Hz");
        }
        else tfmP.printReply();

        Serial.print( "Save settings rate: ");
        if( tfmP.sendCommand( SAVE_SETTINGS, 0, _tfmini_adress))
        {
          Serial.println("");
        }
        else tfmP.printReply();
```

I did not see any error on the serial output, but I know the the framerate was not set accordingly because the TFMini is always getting new measures without triggering new ones (FRAME_0). And I'm also able to see the TFMINIPLus IR led on when the frame rate is 0Hz.

Do you know if this is an issue with the lib or with the TFMini Plus itself ?

Thank you

budryerson, 2021-09-15:
Ronan,<br />
Your "dirty code" is not correct.  Your command `0x5a, 0x06, 0x03, 0xe8, 0x03` makes no sense.  The second byte, Command Length, is `0x06` but the array `cmndData[5]` is only five bytes long.  Please review **7.2 Command Convention** in the Product Manual, p.17.<br />
The fragment of code you sent: `tfmP.sendCommand( SET_FRAME_RATE, FRAME_20, _tfmini_adress)`  is setting the frame rate to 20Hz. Is that what you want?<br />
The TFMini-Plus IR led is always on unless power is disconnected or the output is disabled.  Setting the frame rate to zero does not disable the output.<br />
Q. Do you know if this is an issue with the lib or with the TFMini Plus itself ?
A. It is most likely that the issue is neither the library nor TFMini-Plus.<br  />
Please send me your entire sketch and a sample of your output.<br />
Thanks,
Bud Ryerson
San Francisco

ronangaillard, 2021-09-15:
Thanks for your quick answer @budryerson 

Command is actually 6 bytes length because I need to send the checksum after. (I'm sending exactly the example command from the 7.2 Command Convention in the Product Manual).

> The fragment of code you sent: tfmP.sendCommand( SET_FRAME_RATE, FRAME_20, _tfmini_adress) is setting the frame rate to 20Hz. Is that what you want?

Sorry my bad it's a typo, I tested with `tfmP.sendCommand( SET_FRAME_RATE, FRAME_0, _tfmini_adress)`.

> The TFMini-Plus IR led is always on unless power is disconnected or the output is disabled. Setting the frame rate to zero does not disable the output.

Mmmmm this is very bad for energy impact...

ronangaillard, 2021-09-15:
My bad, I really did a typo in my code and set the framerate to 20 instead of 0. And this works as expected.

(And the IR Led turns off 👍 )

Not sure why my dirty code does not work but I'll move to your library 😉

closed 2021-09-15 by ronangaillard