budryerson/TFMini-Plus-I2C #18: Is it Possible to Measure Relative Speed of an Object?

shivamkk007, 2022-03-22:
Hey Bud,
 Is it possible to measure the relative speed of an object using a single Module?
Using the previous values of distance w.r.t time passed?

shivamkk007, 2022-03-22:
And can AT Mega Handle 4 Sensors on I2C with its own power supply pins?

shivamkk007, 2022-03-23:
AT mega has 2 SDA and SCL pins in which one should I connect?

budryerson, 2022-03-23:
You asked "How can I do that?"  The simple answer is: distance / time = speed

Your above output image suggests that the sketch is repeatedly trying to reconnect with the I2C device.  Use an I2C scanner/detector sketch to make sure that the device is connected properly.

I believe the Mega 2560 R3 has the same I2C pins on two different locations.  You should use pins 20 and 21.

shivamkk007, 2022-04-04:
Hey Bud,
I have checked all the connections, in the IWC changing code the Devices are visible but when I run the main Example code the same 00:00:00:00 output is shown. How can I fix that.

shivamkk007, 2022-04-07:
Here is the code -
`
#include <Wire.h>     // Arduino standard I2C/Two-Wire Library
#ifndef __PRINTF_H__
#define __PRINTF_H__

#if defined (ARDUINO) && !defined (__arm__) && !defined(__ARDUINO_X86__)

int serial_putc( char c, FILE * )
{
  Serial.write( c );

  return c;
}

void printf_begin(void)
{
  fdevopen( &serial_putc, 0 );
}

#elif defined (__arm__)

void printf_begin(void){}

#elif defined(__ARDUINO_X86__)
int serial_putc( char c, FILE * )
{
  Serial.write( c );

  return c;
}

void printf_begin(void)
{
  //For reddirect stdout to /dev/ttyGS0 (Serial Monitor port)
  stdout = freopen("/dev/ttyGS0","w",stdout);
  delay(500);
  printf("redirecting to Serial...");
  
  // -----------------------------------------------------------
}
#else
#error This example is only for use on Arduino.
#endif // ARDUINO

#endif // __PRINTF_H__  // Modified to support Intel based Arduino
                      // devices such as the Galileo. Download from:
                      // https://github.com/spaniakos/AES/blob/master/printf.h

#include <TFMPI2C.h>  // TFMini-Plus I2C Library v1.7.3
TFMPI2C tfmP;         // Create a TFMini-Plus I2C object

void setup()
{
    Serial.begin( 115200);   // Initialize terminal serial port
    printf_begin();          // Initialize printf library.
    delay(20);
    
    printf("\n");            // say 'hello'
    printf( "TFMPlus I2C Library Example - 14JAN2022");
    printf("\n\n");

    // - - - - -   RECOVER I2C BUS  - - - - - - - - - - - - - - - 
    // An I2C device that quits unexpectedly can leave the I2C bus hung,
    // waiting for a transfer to finish.  This function bypasses the Wire
    // library and sends 8 pseudo clock cycles, a NAK, and a STOP signal
    // to the SDA and SCL pin numbers. It flushes any I2C data transfer
    // that may have been in progress, and ends by calling `Wire.begin()`.
    tfmP.recoverI2CBus();

     Wire.begin();            // Called in previous function.
    Wire.setClock( 400000);  // Set I2C bus speed to 'Fast' if
                                 // your Arduino supports 400KHz.

    // Send some example commands to the TFMini-Plus
    // - - Perform a system reset - - - - - - - - - - -
    printf( "System reset: ");
    if( tfmP.sendCommand( SOFT_RESET, 0))
    {
        printf( "passed.\r\n");
    }
    else tfmP.printReply();  // This response and 'printStatus()' are for
                             // troubleshooting and not strictly necessary.
    //
    // - - Display the firmware version - - - - - - - - -
    printf( "Firmware version: ");
    if( tfmP.sendCommand( GET_FIRMWARE_VERSION, 0))
    {
        printf( "%1u.",  tfmP.version[ 0]); // print three single numbers
        printf( "%1u.",  tfmP.version[ 1]); // each separated by a dot
        printf( "%1u\n", tfmP.version[ 2]);
    }
    else tfmP.printReply();
    //
    // - - Set the data frame-rate to 20 - - - - - - - - -
    printf( "Data-Frame rate: ");
    if( tfmP.sendCommand( SET_FRAME_RATE, FRAME_20))
    {
        printf( "%2uHz.\n", FRAME_20);
    }
    else tfmP.printReply();
    // - - - - -   End of example commands- - - - - - - - - -

/*  // - - - - - - - - - - - - - - - - - - - - - - - -
    // The next two commands may be used to switch the device to
    // UART (serial) mode.  If used, this sketch will no longer
    // receive I2C data.  The 'TFMPlus_example' sketch in the TFMPlus
    // (serial) Library can be used to switch the device back to
    // I2C mode.   Don't forget to switch the cables, too.
    // - - - - - - - - - - - - - - - - - - - - - - - -
    // - - Set Serial Mode - - - - - - - - - - -
    printf( "Set Serial Mode: ");
    if( tfmP.sendCommand( SET_SERIAL_MODE, 0))
    {
        printf( "mode set.\r\n");
    }
    else tfmP.printReply();
    printf( "Save Settings: ");
    if( tfmP.sendCommand( SAVE_SETTINGS, 0))
    {
        printf( "saved.\r\n");
    }
    else tfmP.printReply();
*/

    delay(500);            // And wait for half a second.
}

// Initialize data variables
int16_t tfDist = 0;       // Distance to object in centimeters
int16_t tfFlux = 0;       // Signal strength or quality of return signal
int16_t tfTemp = 0;       // Internal temperature of Lidar sensor chip

// = = = = = = = = = =  MAIN LOOP  = = = = = = = = = =
void loop()
{
    tfmP.getData( tfDist, tfFlux, tfTemp); // Get a frame of data
    if( tfmP.status == TFMP_READY)         // If no error...
    {
        printf( "Dist:%04icm ", tfDist);   // display distance,
        printf( "Flux:%05i ", tfFlux);     // display signal strength/quality,
        printf( "Temp:%2i%s", tfTemp, "°C" );   // display temperature,
        printf( "\n");                     // end-of-line.
    }
    else
    {
        tfmP.printFrame();                 // Display error and data frame
        if( tfmP.status == TFMP_I2CWRITE)  // If I2C error...
        {
            tfmP.recoverI2CBus();          // recover hung bus.
        }
    }
    delay(50);    //  Run loop at approximately 20Hz.
}`

The O/P-

[image]

The Setup-

[image]

[image]

[image]

budryerson, 2022-04-07:
1) Please send your sketch and your libraries as folders and your output as a text file.
2) Please connect only one LiDAR device at a time until you get this working.
Thanks,
Bud

shivamkk007, 2022-04-10:
Where should I send the Files? github isn't accepting the zip file maybe because of its size(around 28Mb).

budryerson, 2022-04-10:
That won't fit through email either.  Put it up on DropBox (or a file hosting service of your choice) and send me a link.  I cannot wait to see what you are sending. Twenty-eight megabytes.  Wow!
Bud

shivamkk007, 2022-04-10:
Here is the Link 
[https://drive.google.com/drive/folders/1EMkMi4ePw2QsqFOGZipd2iVlmy8046y8?usp=sharing](url)
I am sure you'd be disappointed.
Shivam

budryerson, 2022-04-10:
On April 4th, you wrote, "while changing the I2C, the Devices are visible."  Please send me a text file of your output from the `TFMPI2C_changeI2C` sketch.  If you used a different sketch to change the I2C address, please send that sketch along with a file of the output.
Thanks,
Bud

shivamkk007, 2022-04-10:
Hey Bud,
 I have updated the same link with I2C sketch, Output and Setup Pictures.

budryerson, 2022-04-10:
The sketch says: `00:13:26.734 -> No I2C devices found.`   Are the "Devices" in I2C communications mode?  Are they connected correctly?  How did you set their I2C slave addresses?

shivamkk007, 2022-04-11:
[image]

Yes they are connected in I2C mode. But I am getting some random devices as you may see in the Output later on 
00:13:50.943 -> First I2C address found: 8 (0x08 Hex)
00:14:25.545 -> I2C device found at address 110 (0x6E Hex)
00:14:41.541 -> First I2C address found: 56 (0x38 Hex)

budryerson, 2022-04-11:
1) Your device is randomly and intermittently changing addresses.
2) Your device is not connected and your Arduino is looking at noise.
3) Your device is still in serial mode.
Number 3 is most likely to me.  If you have an oscilloscope you can examine the signal.  Or you can hook it up to the GUI again. and see if you can still communicate in serial mode.  Let me know what you find out.
Good luck,
Bud

shivamkk007, 2022-04-17:
Isn't there a need of providing pull up resistance and voltage supply to the sensor other than the power input pins?

budryerson, 2022-04-17:
No.  In all my experience using a Mega2560 I have never needed nor used pull-up resistors.
I have powered a single TFMPlus from a Mega2560; but, as I wrote earlier, I recommend that you power all devices separately and independently from the processor.
Are the devices in I2C mode? Are they connected correctly? How did you set their addresses?
Please be kind enough to answer my questions.
Thank you,
Bud

shivamkk007, 2022-04-18:
Finally I was able to set a sensor on I2C mode.

[image]

shivamkk007, 2022-04-18:
Hey Bud,
I have set 2 of the TFMini plus on I2C with different address.
How can I take input from both?

shivamkk007, 2022-04-19:
> Hey Bud,
> I have set 2 of the TFMini plus on I2C with different address.
> How can I take input from both? Also how can I use the past values of distance in order to calculate the relative speed?

I have figured out how to use multiple sensors on I2C one of which is TF02 Pro, currently using 3 sensors 2 TFMiniP and a TF02 Pro. The main problem is now the use of past value of Distance with respect to Time. How can I achieve that?

budryerson, 2022-04-21:
"I have set 2 of the TFMini plus on I2C with different address. How can I take input from both?"
Use the library's command `getData( distance, address);` where `address` is the address of your slave device.

"I have figured out how to use multiple sensors on I2C one of which is TF02 Pro,"
What library do you use for the TF02 Pro?

"The main problem is now the use of past value of Distance with respect to Time. How can I achieve that?"
Save the past value of Distance in memory.  Wait.  Recall the value at a later Time.  Problem solved!

closed 2022-04-21 by budryerson