budryerson/TFMini-Plus-I2C #19: Trouble with Tfmini-s and Arduino 33 Iot

Dariog86, 2022-03-25:
Hi Bud.
I'm trying to use the Tfmini-s with arduino 33 iot, but something strange happened.
I've successfully set the sensor in I2C mode, and it is recognized by arduino with address 0x10 (i've used a simple I2c Scanning sketch to verify the connection).
When I upload the TFMPI2C_example sketch into arduino, on the serial monitor nothing is shown as you can see on image 1.
[image]

I've tried to uncomment the line 88 amd 89 and some data appears as in image 2
[image]

but none of the message appears on the serial monitor, not the initial message nor the value of mesurement.
i don't know if it's a problem with this particular model of arduino.
I've tested the sensor with the GUI and the data were displayed correctly

thanks in advance for your reply

budryerson, 2022-03-25:
That is the first message you should receive from the sketch, before anything else, is the greeting: `TFMPlus I2C Library Example - 14JAN2022`.  I don't see that.
Please bundle up all of your code, including all the libraries you are using, a text file of your output and maybe some pictures of your setup so I can see your wiring into a zip file and send it to me.
I'm sure we can fix this.

Bud Ryerson
San Francisco

Dariog86, 2022-03-26:
Correct. None of the message are displayed.
I've made a zip file with all you requested with some text file as notes.
i'm happy that you say that we can fix the issue.
Thanks
[code_log_setup.zip](https://github.com/budryerson/TFMini-Plus-I2C/files/8355250/code_log_setup.zip)

budryerson, 2022-03-26:
I think we might be having a problem with `printf.h`.  The Arduino IDE may include it's own version of `printf.h` in the nano compiler.  Trying commenting out line #63 of the "untouched" example.

> `// #include "printf.h"   // Modified to support Intel based Arduino`

An alternative is to replace all the `printf()` statements with `Serial.print()`s.  That's not a big problem, but let's try my first suggestions first.

Dariog86, 2022-03-26:
Unfortunately no luck Bud. 
I've commented out line 63 and line 73 on ` printf_begin();` because of initialization of printf.
Unfotunately I've still no output on the serial monitor. It's completely blank unfortunately.
I've tried to uncomment the same lines on the other sketch with line 88 and 89 uncommented  but the result is the same as previous with all the line Status: CHECKSUM Data etc...

budryerson, 2022-03-27:
Did you comment out `printf_begin()` because you got an error?  I would leave it;, but I'm not in the lab, and I could be wrong.
Anyway, let's go to Plan B.  Replace `printf()` with `Serial.print()`.  In some places, it this will require more than one statement.
Good luck.

Dariog86, 2022-03-27:
We made a step ahead !
these are the steps i made to have some progress.
1 - I've replaced `printf()` to `Serial.print()`
2 - load sketch on arduino and look for serial monitor, no initial message and still Status: CHECKSUM Data etc...
3 - i noticed an output mesurement if i completely cover the sensor with my hand (you can see it in attached log around 16:58:45)
4 - i've disconnected the sensor connector from the sensor itself (as you can see in log around 16:58:52 when are searching for I2C)
5 - after reconnected the sensor to its cable i've got the data outputted (16:58:54 on the log).

I've tried to reboot arduino and i needed to redo these step to get it working, and i noticed that isn't unnecessary to detach the whole cable from the sensor, but only one of the 2 cables that goes in pin 4 and 5 (SDA and SCL) to get it work.

It seems that arduino or the code doesent initialize correctly the communication, and doing a "manual reset" detaching one of the 2 cables start the communications. 
In this case it make sense that the initial message wasn't showed, because the sensor is not "initialized".

I don't know if i can explain it correctly, and i doesen't have so many electronics foundamentals, but in my mind if there was a method to do this "initialization" by code it will fix the issue.
[code_and_log.zip](https://github.com/budryerson/TFMini-Plus-I2C/files/8357946/code_and_log.zip)

budryerson, 2022-04-04:
May I see your updated code and output, please?
Also, you should comment out  `Wire.setClock( 400000);` and not push the I2C bus until everything else is working.

Dariog86, 2022-04-04:
Hi Bud.
This week end i've worked on the code and i've found a solution !
After some attemps, adding `Wire.begin();` just before `tfmP.printFrame();` on the main loop (line 170 in the attached code)
make all the system work.
Initial message, firmware version etc. and after that immediately start to print the correct data.
It seems that,in this case, the `tfmP.recoverI2CBus();` function doesn't do the `Wire.begin();`.

[TFMPI2C.zip](https://github.com/budryerson/TFMini-Plus-I2C/files/8411623/TFMPI2C.zip)

budryerson, 2022-04-05:
Thank you for the updated code.  I am happy to hear that you're making progress; but I am a little mystified by your explanation.  To help me understand, may I see your output, please?
Thanks,
Bud

Dariog86, 2022-04-06:
Sure
Thanks
[output.txt](https://github.com/budryerson/TFMini-Plus-I2C/files/8429127/output.txt)

budryerson, 2022-04-06:
The Arduino 33 IoT uses the Arm SAMD21 core.  For that, the Arduino IDE uses some specialized libraries, usually located here:
`C:\Users\**your user name**\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\libraries`
Among them is a different version of the Wire library, which might help to explain some things.

I disagree with some choices you made (why call `Wire.begin()` so often, why "Fast" mode, etc); but I am happy to hear that your code is working.  It is exciting that you get to play with new technology.  I hope you will continue working to improve your code, and that as you do you will keep us all informed.  I may get one of these to play with myself.

All the best,

Bud

closed 2022-07-14 by budryerson

gcsalzburg, 2023-05-11:
Just to add - I'm also testing this library on an Feather M0 which uses the same SAMD21 core.
I followed the steps to replace all printf() calls with Serial.print() calls as suggested by Bud here: https://github.com/budryerson/TFMini-Plus-I2C/issues/19#issuecomment-1079809775

This fix seems to work well, no need for the multiple Wire.begin() commands and other such experiments :)