budryerson/TFMini-Plus-I2C #21: The Benewake GUI works only in Serial Mode.  If you sent a command to change to I2C mode, and the command was successful, then you can only communicate in I2C mode.  An Arduino Pro Mini or Mega 2560 is not expensive and has an I2C interface.  If you get an Arduino, I can help you.  We will make this work.

segevoved, 2022-07-13:
The Benewake GUI works only in Serial Mode.  If you sent a command to change to I2C mode, and the command was successful, then you can only communicate in I2C mode.  An Arduino Pro Mini or Mega 2560 is not expensive and has an I2C interface.  If you get an Arduino, I can help you.  We will make this work.

_Originally posted by @budryerson in https://github.com/budryerson/TFMini-Plus-I2C/issues/11#issuecomment-772266615_

budryerson, 2022-07-14:
Sure.  It's easy.  But you didn't tell me which Benewake device you are trying to use.  And you didn't tell me what you've tried so far.  Let me know and we'll fix this.

segevoved, 2022-07-14:
Hey! I got the LIDARS to work. I'm using 4 TFluna, and 1 TF03 (down)
I have a cube orange and I try to set 5 lidars to work with it. (front, back, left, right, down).

At the start, I tried using Terrabae M15 but for some reason, that configuration only seemed to be working on cube black. (I think there is a problem with the H7 architecture and I2C).

Anyway, I ended up trying to use TFluna and TFmini that I had lying around.
And now, I only got the TFluna to work using an Arduino UNO to check their addresses.
Honestly, I prefer using the TFmini since it has a longer range (12m). If you could help me with getting it to work that would be amazing.
And another thing. I got the UART version of the TF03 and except for TX-RX, it has CAN-H and CAN-L cables. 
Can I use that to connect to the CAN2 port? all other ports are occupied at the moment. I'm not sure what to configure to get it to work in parallel with the other lidars (right, left, front, back).

segevoved, 2022-07-19:
Hey, an update. I managed to make it work, used an Arduino to read the addresses of the TFluna, and even the TF03 works in the CAN port.
And now, I have a new problem.

I want the drone to stop in AUTO mode (Guided, Mission) but when I'm sending it on a mission or telling it to go to a point on the map it won't listen to the rangefinders, If there's a wall it will just crash. 

How can I make it stop when it recognises an object in AUTO mode?