budryerson/TFMini-Plus-I2C #24: Possible Bug (Incorrect distance measurement)

Firestarter97, 2022-09-11:
Hello,

There may be a bug in the library. I've been testing out the tf mini plus I2C version and I've noticed once in a while the sensor gives me an output distance value saying that's essentially saying there's an object close to it when in reality that isn't the case. I did separate the SDA and SCL wires as much as possible. The distance value that it gives me is 0 cm. Any way to further debug this?

Thanks

Firestarter97, 2022-09-11:
I created a workaround for it by essentially saying that if the measured distance is equal to 0 or 3cm then we just ignore it. But wondering if there's something else that could be done?

24gdolphin, 2022-09-29:
Hi，The manual say the module some registers need to be when in a small distance, You can find more details in manual, Best regards.
[image]