budryerson/TFMini-Plus-I2C #25: Can't get TFMini Plus to show MM

TahoePaul, 2023-02-12:
First of all, I'm a rank beginner... but I'm amazed at what I've been able to accomplish so far.  At any rate, it's possible that I'm missing something very obvious.  :/  

I'm trying to get my TFMini Plus to return results in MM, rather than CM, and have had no luck with the I2C_FORMAT_MM or STANDARD_FORMAT_MM commands.  (I'm running in I2C mode, so I'm thinking that I should be using the I2C_FORMAT_MM command.)

I'm including both of these statements in the setup section of my code:

  if (tfmP.sendCommand(I2C_FORMAT_MM, 0)) {     
    Serial.print("I2C Format set to MM ");      
  } else tfmP.printReply();                     

  if (tfmP.sendCommand(SAVE_SETTINGS, 0)) {     
    printf("Settings saved.\r\n");               
  } else tfmP.printReply();                     

Both commands are being run and are not falling though to the else part of the if statements.  I continue to see returned results in the CM format and can't figure out why.  

Any help you can provide would be greatly appreciated!  

I'm building a pump controller module to control a pump in a sump pit... I should say that I'm not really sure that I'd gain any granularity by switching to MM.  I think the TFMini Plus has a granularity of 5mm but I'm only seeing a granularity of 1CM when I'm in CM mode.  (This is the reason I'm trying to switch over to MM mode.)

Thank you.

TahoePaul, 2023-02-12:
Here's  a view of the code I'm trying to get working... simplified into one of the example sketches (Hopefully this formats correctly.):

#include <Wire.h>    // Arduino standard I2C/Two-Wire Library
#include "printf.h"  // Modified to support Intel based Arduino
                     // devices such as the Galileo. Download from:
                     // https://github.com/spaniakos/AES/blob/master/printf.h

#include <TFMPI2C.h>  // TFMini-Plus I2C Library v1.7.3
TFMPI2C tfmP;         // Create a TFMini-Plus I2C object

void setup() {
  Serial.begin(115200);  // Initialize terminal serial port
  printf_begin();        // Initialize printf library.
  delay(20);

  printf("\n");  // say 'hello'
  printf("TFMPlus I2C Library Example - 14JAN2022");
  printf("\n\n");

  // - - - - -   RECOVER I2C BUS  - - - - - - - - - - - - - - -
  // An I2C device that quits unexpectedly can leave the I2C bus hung,
  // waiting for a transfer to finish.  This function bypasses the Wire
  // library and sends 8 pseudo clock cycles, a NAK, and a STOP signal
  // to the SDA and SCL pin numbers. It flushes any I2C data transfer
  // that may have been in progress, and ends by calling `Wire.begin()`.
  tfmP.recoverI2CBus();

  //  Wire.begin();            // Called in previous function.
  //  Wire.setClock( 400000);  // Set I2C bus speed to 'Fast' if your Arduino supports 400KHz.

  // Send some example commands to the TFMini-Plus
  if (tfmP.sendCommand(I2C_FORMAT_MM, 0)) {       //****************
    printf("I2C Format set to MM. ");                            //****************
  } else tfmP.printReply();                                            //*****************
  delay(1000);                                                             //  So I can see if it's working.
  //
  if (tfmP.sendCommand(SAVE_SETTINGS, 0)) {         //*****************
    printf("Settings saved.\r\n");                                  //******************
  } else tfmP.printReply();                                           //******************
  delay(1000);                                                            //  So I can see if it's working.
  // - - Display the firmware version - - - - - - - - -
  printf("Firmware version: ");
  if (tfmP.sendCommand(GET_FIRMWARE_VERSION, 0)) {
    printf("%1u.", tfmP.version[0]);  // print three single numbers
    printf("%1u.", tfmP.version[1]);  // each separated by a dot
    printf("%1u\n", tfmP.version[2]);
  } else tfmP.printReply();
  //// - - - - -   End of example commands- - - - - - - - - -

  delay(500);  // And wait for half a second.
}

// Initialize data variables
int16_t tfDist = 0;  // Distance to object in centimeters
int16_t tfFlux = 0;  // Signal strength or quality of return signal
int16_t tfTemp = 0;  // Internal temperature of Lidar sensor chip

// = = = = = = = = = =  MAIN LOOP  = = = = = = = = = =
void loop() {
  tfmP.getData(tfDist, tfFlux, tfTemp);  // Get a frame of data
  if (tfmP.status == TFMP_READY)         // If no error...
  {
    printf("Dist:%04icm ", tfDist);      // display distance,
    printf("Flux:%05i ", tfFlux);        // display signal strength/quality,
    printf("Temp:%2i%s", tfTemp, "°C");  // display temperature,
    printf("\n");                        // end-of-line.
    printf("Dist:%04i\n\n", tfDist);       //  (MY CODE)        **********  I'm trying to get MM here.
  
  } else {
    tfmP.printFrame();                 // Display error and data frame
    if (tfmP.status == TFMP_I2CWRITE)  // If I2C error...
    {
      tfmP.recoverI2CBus();  // recover hung bus.
    }
  }
  delay(200);  //  Run loop at approximately 20Hz.
}
// = = = = = = = = =  End of Main Loop  = = = = = = = = =

budryerson, 2023-02-12:
Because the accuracy of the device is ±50mm, the library `getData `command is coded to ask for a return **only** in CM (see below).  So even if you get a MM return, it will not be accurate.  But I understand your desire.

Your use of the `I2C_FORMAT_MM` command is not correct.  If you were in serial mode, you would use the `STANDARD_FORMAT_MM` command to format the output data stream.  At this time, the `I2C_FORMAT_MM` command is not available to you through the library.  I should fix that.

I cannot get to the lab to try things out for a couple of days.  In the meantime, I could send you a patched library that asks for data in MM. Or you could patch it yourself.  In **TFMPI2C.cpp**, down around line 97, change the `I2C_FORMAT_CM` to `I2C_FORMAT_MM` and see what that does. Let me know... and good luck.
```
    // - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    // Step 0 - Command device to ready distance data in centimeters
    // - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    // The device can also return data in millimeters, but its
    // resolution is only 5mm (0.5cm) and its accuracy is ±5cm.
    if( sendCommand( I2C_FORMAT_CM, 0, addr) != true) return false;
```

TahoePaul, 2023-02-12:
Thank you very much for your reply... I'll get back to you on this.  I think I can "fix" things with what you told me in your message.  :)

Also, I'm not so much interested in "accuracy" as in "repeatability," and a fairly small "granularity."  (That way I can write my program in such a way as to turn my pump on, and off, based on the experienced surface level of the water.)  

Anyway... I'm off to edit the TFMPI2C.cpp file now.  I had seen that code while exploring but, well, don't know enough to go and start making changes on my own.  

I'll let you know how things go...

TahoePaul, 2023-02-12:
Looks like I'm now getting MM as a return value.  Very exciting!  I'm off and running again!  Thank you VERY much for your help!

closed 2023-02-12 by TahoePaul