budryerson/TFMini-Plus-I2C #26: Compatible with TF02-Pro

gcsalzburg, 2023-04-14:
Hi, 

Do you know if this library is compatible with the TF02-Pro?
It was working successfully in UART mode (default out of the box) and I tried to use the library to test the I2C mode and it didn't work. Specifically, I just see the following Serial Monitor output:

```
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
 Status: I2C-WRITE Data: 00 00 00 00 00 00 00 00 00
Recover default I2C bus.
```
Having tested this, the device no longer works in UART mode with my original test script either!

Any suggestions are much appreciated.

George

budryerson, 2023-04-15:
George,
I'm sorry to hear you're having problems with your TF02-Pro.
You asked if the library was compatible.  Benewake says that is.  I don't have a TF02-Pro to play with.
I have many questions.
•   First, did you initially, successfully use the TFMini-Plus library while in UART mode?
•   Then, did you use the TFMini-Plus library to command the device to I2C mode?
•   Finally, have you used the `TFMPI2C_changeI2C.ino` sketch in the `examples` folder or any other I2C scanner sketch to check the I2C address of the device?
Please let me know, and please be very thorough in your answers.
Bud Ryerson
San Francisco

gcsalzburg, 2023-04-17:
Thank you Bud - really appreciate the quick response. No, I had not tried these steps in that order no, and you've made me realise I might have missed an important step.
I'll try this out and get back to you very soon with the detailed answers :)

gcsalzburg, 2023-05-11:
Hi Bud,
After some time away I was able to get back to this and I'm pleased to confirm that your instructions were perfect and everything works now.
I had most likely failed to set the device to I2C mode correctly. For completeness:
1. I successfully used the TFMini-Plus library in UART mode
2. I successfully ran the command to change to I2C mode
3. I used the I2C scanner to identify the I2c address (= 16, 0x10 Hex)
4. I could then use the TFMPI2C library successfully.

Thanks for your prompt response a few weeks ago and your great effort with this library and others.
Happy to report this library is TF02-Pro compatible out of the box :)

George