budryerson/TFMini-Plus-I2C #30: Sorry for being a Noob, help needed on how to use this please

stevefoxuk, 2025-07-05:
I own 3 TF Mini Plus units I used the Official GUI to change the slave i2c address on two (or so I thought) and set all to use i2c and so now I cannot communicate with them via the official GUI.

One correctly changed its i2c address to 17 and the other was meant to be changed to 18 but remains on the default 16 which clashes with the one I left on the default address so now I have come across this and trying to figure out how to use it to change the address to 18.

As mentioned above, I have the TF Mini Plus, I have uploaded your program to an Arduino nano.

Where are commands entered, are they done through IDE serial monitor or are they configured in the sketch itself?

In serial monitor, I just see garbled characters.

I have tried to figure this out before asking but im no coder and well out my depth.

Help would be massively appreciated.

Thanks, Steve.

budryerson, 2025-07-05:
You beat me too it!  If you get a chance, I'd love to know more about your project.  Is there a website? Or even just a picture would be great.

Bud

stevefoxuk, 2025-07-24:
.